openmovementproject/openmovement #4: Example code used to convert CWA-DATA.CWA

imochaz, 2015-02-26:
Hi,

I tried your example code for converting a CWA-DATA.CWA file in the device into a CSV file, but I got different results from OpenMovement software.

Could you please give an example code in either MATLAB or Python to do the same conversion as in OpenMovement Software? Also, which code you use to convert from timestamp into datetime string?

Thanks,
Akara

danielgjackson, 2015-03-24:
Conversion utilities are included in the OMGUI software:
  https://github.com/digitalinteraction/openmovement/wiki/AX3-GUI

Conversion source code is available in "Software/AX3/cwa-convert":
  https://github.com/digitalinteraction/openmovement/tree/master/Software/AX3/cwa-convert

The Python code is probably the least used/tested. If you have used code that outputs raw sensor values (they will all be integers), you must first divide by 256 to convert it to units of 'g'.

closed 2015-03-24 by danielgjackson

imochaz, 2015-03-24:
I have tried your source code

https://github.com/digitalinteraction/openmovement/tree/master/Software/AX3/cwa-convert

before I posted this issue, but the problem is the output of OMGUI software

https://github.com/digitalinteraction/openmovement/wiki/AX3-GUI

is different from the output of MATLAB, python, Java, etc.

danielgjackson, 2015-03-24:
Hi Imochaz,

You will have to be more specific -- what differences are you seeing, and between exactly which methods?  The on-device ("raw") data from a .CWA file is subject to the accelerometer's true sampling rate (which can vary gradually over time).  The GUI can convert this raw data as it stands, or can re-interpolate this to a fixed frequency (these will give slightly different results). Other importers are usually only for this raw (un-interpolated) data, but may be presenting the underlying sensor units (if they are integers) and will need dividing by 256 to units of 'g'; and will almost certainly have differences in how they choose to interpolate over the periodic timestamps. Yet others importers (e.g. the Python version) are not well used or tested, and should not be relied upon.

Cheers,

Dan