openmovementproject/openmovement #6: Split out repository

Woodz, 2015-03-16:
Hello,

I am trying to fork a copy of your Java code for converting .CWA files. However, as this repository contains all your software sources and releases, I cannot just clone this single folder. This makes working with your repository and code difficult. I recommend you try to follow GitHub recommended behaviour (see https://guides.github.com/activities/hello-world/) and have a separate repository for each piece of software. This will allow you to individually version and release components instead of the entire repository as you currently have.

Cheers

danielgjackson, 2015-09-11:
Yes, unfortunately, this structure was inherited from the old codebase at Google Code -- we plan to do this in the near future.

danielgjackson, 2021-07-14:
Some parts are split out, and associate repositories include:

* OMAPI: https://github.com/digitalinteraction/libomapi/
* Python: https://github.com/digitalinteraction/openmovement-python/
* omsynth: https://github.com/digitalinteraction/omsynth/
* omconvert: https://github.com/digitalinteraction/omconvert/
* timesync: https://github.com/digitalinteraction/timesync/
* azimuth: https://github.com/digitalinteraction/azimuth/

...the rest probably won't be split in the near future, so closing the issue for now.

closed 2021-07-14 by danielgjackson