openmovementproject/openmovement #7: cwa-convert Java - CwaBlock copy constructor misses fields

Woodz, 2015-03-24:
The copy constructor at https://github.com/digitalinteraction/openmovement/blob/master/Software/AX3/cwa-convert/java/src/newcastle/cwa/CwaBlock.java#L130 is incorrect as it only copies some of the fields. The temperature, battery, light, events and number of samples are left as the default 0 value. This causes issues particularly in the method `CwaReader.readBlock` at https://github.com/digitalinteraction/openmovement/blob/master/Software/AX3/cwa-convert/java/src/newcastle/cwa/CwaReader.java#L116 that uses the copy constructor to return a copy of the current block. Due to this bug, the block returned is missing key fields so cannot be used.

danielgjackson, 2015-03-24:
Thanks for the bug report -- the copy constructor has now been fixed. The Java code has not been used very heavily before, so this issue had gone unnoticed.

closed 2015-03-24 by danielgjackson

Woodz, 2015-03-25:
Thank you for your prompt response and fix!