openmovementproject/openmovement #11: OMGui - Temperature calibration is incorrect

Woodz, 2015-03-26:
The calibration applied to the temperature reading in the OMGui application is `temperature * 75000 / 256 - 50000` to give the battery level in millicentrigrade (https://github.com/digitalinteraction/openmovement/blob/master/Software/OM/omapi/src/omapi-reader.c#L549). On my test file, this displays as 35.25^C in the UI, which is too hot.

Compare this to the calibration specified in the Matlab cwa-convert application of `temperature * 150 - 20500)/1000` to give the battery level in degrees celcius (https://github.com/digitalinteraction/openmovement/blob/master/Software/Analysis/Matlab/AX3_readFile.m#L250) which on the same test data set gives a temperature of 23.15C, which is much more realistic.

From this comparison, it appears as though the OMGui application temperature calibration is incorrect.

chanshing, 2023-06-27:
There seems to be conflicting documentation on the correct formula to use. In addition to those found by @Woodz , apparently there are also:
-  `temperature * 75.0f / 256 - 50` https://github.com/digitalinteraction/openmovement/blob/545564d3bf45fc19914de1ad1523ed86538cfe5e/Software/AX3/cwa-convert/c/main.c#L830
- `temperature * 19.0 / 64.0 - 50.0` https://github.com/digitalinteraction/openmovement/blob/545564d3bf45fc19914de1ad1523ed86538cfe5e/Software/AX3/cwa-convert/c/main.c#L720

Furthermore, there is documentation showing two more formulas (one of them is probably a typo) which, confusingly, I haven't seen them applied anywhere in the codebase:

- https://axivity.com/help/2 
[image]

- https://axivity.com/userguides/ax3/settings/
[image]

In our Biobank Accelerometer Analytics tool https://github.com/OxWearables/biobankAccelerometerAnalysis we use the second formula that @Woodz mentioned, and this is the formula that has been used for many published UK-Biobank papers already.

@danielgjackson @aidendoherty

danielgjackson, 2023-07-05:
The AX devices use an MCP9700 sensor for temperature, and the correct temperature conversion from ADC values is:

> $temperature = adcValue * 75 / 256 - 50$

...for derivation, see [AX3 Auxiliary Data - Device Temperature](https://github.com/digitalinteraction/openmovement/blob/master/Docs/ax3/ax3-auxiliary.md#device-temperature).

This is the same, after scaling for milli-C, as:

* $value * 75000 / 256 - 50000$

For a standard range of temperatures, this is close (within 1°C) to the over-simplified approximation:

* $value * 19 / 64 - 50$

...and within about 0.1°C of the decimal approximation:

* $(value - 171) / 3.413$

...but not very similar (around 1-2°C) to this version (seemingly mistakenly typed as an approximation of Pi):

* $(value - 171) / 3.142$

---

However, the other formula:

* $(value * 150 - 20500)/1000$

...is for the *MCP9701* sensor, which is in the same *MCP970X* family, and shares the [same datasheet](https://ww1.microchip.com/downloads/aemDocuments/documents/MSLD/ProductDocuments/DataSheets/MCP9700-Family-Data-Sheet-DS20001942.pdf), but uses slightly different scale and offsets.

This is a mistake, and this sensor is not used in the AX devices.

This does affect the primary use of the internal temperature sensor -- accelerometer self-calibration -- as the conversions are both linear transformations (as too is the raw ADC value) and the self-calibration method will work the same.  Relative changes will have a scaling factor from the true internal temperature ($20/39$).

However, this will affect absolute readings, for example:

Actual (&deg;C) | Mis-reported (&deg;C) 
-- | --
10.9 | 10.7
15.6 | 13.1
20.3 | 15.5
25.0 | 17.9
29.7 | 20.3

The mis-reported internal temperature can be converted to a true internal temperature as follows:

* $temperature = (misReportedTemperature * 250 / 128) - (1275 / 128)$

chanshing, 2023-07-06:
Thank you so much @danielgjackson 
Very clear!

closed 2023-09-06 by danielgjackson