openmovementproject/openmovement #12: Document CWA format and calibrations

Woodz, 2015-03-26:
When trying to use the OpenMovement software on an AX3 CWA file, I am finding that the output of the different language converters are giving me different answers. For example, see issue #11 . It would be sensible and appropriate to document the CWA file format in a language-agnostic fashion with the documented calibration factors for all the fields (accelerometer, battery, light, temperature, etc.). In addition, providing a sample CWA file with the equivalent decoded data will be very useful for validating any libraries that claim to read CWA files.

danielgjackson, 2016-09-16:
Now located at:

https://raw.githubusercontent.com/digitalinteraction/openmovement/master/Docs/cwa/cwa-format.txt

A sample file is under consideration.

closed 2016-09-16 by danielgjackson

danielgjackson, 2016-10-18:
Moved to https://raw.githubusercontent.com/digitalinteraction/openmovement/master/Docs/cwa/cwa.h
See also https://raw.githubusercontent.com/digitalinteraction/openmovement/master/Docs/cwa/ax3-technical.md