openmovementproject/openmovement #13: OMGui - timestamps in data viewer are incorrect

Woodz, 2015-03-26:
As shown in the screenshot attached, the timestamps on the right of the graph are before the timestamps on the left of the graph.

Woodz, 2015-03-26:
[image]

danielgjackson, 2016-09-16:
This is related to how the OMGUI preview rapidly finds segments to show without needing to load all of the data from the device/file.

It only happens when very zoomed in (at the interface between between the "overview" zoom level of one column per data sector, and the "detailed" zoom level, where the data within the sector is shown).

It should be fixed (but not an immediate priority due to the limited scope).