openmovementproject/openmovement #17: OSC mode Wax9: allows user to select different ranges for accelerometer, gyro and mag.

12deadpixels, 2015-12-22:
The OSC mode for the Wax9 overrides the user's parameters for acc, gyro and mag ranges.

In waxrec.c the default values for a 8g, 2000 dps conversion are fixed, therefore these cannot be changed in the terminal (only in the OSC mode).

> Waxrec.c line 1441

````
        o += write_osc_float(buffer + o, wax9Packet->accel.x / 4096.0f);        /* [OSC float] Accel. X-axis <4 bytes> */
        o += write_osc_float(buffer + o, wax9Packet->accel.y / 4096.0f);        /* [OSC float] Accel. Y-axis <4 bytes> */
        o += write_osc_float(buffer + o, wax9Packet->accel.z / 4096.0f);        /* [OSC float] Accel. Z-axis <4 bytes> */
        o += write_osc_float(buffer + o, wax9Packet->gyro.x * 0.07f);           /* [OSC float] Gyro.  X-axis <4 bytes> */
        o += write_osc_float(buffer + o, wax9Packet->gyro.y * 0.07f);           /* [OSC float] Gyro.  Y-axis <4 bytes> */
        o += write_osc_float(buffer + o, wax9Packet->gyro.z * 0.07f);           /* [OSC float] Gyro.  Z-axis <4 bytes> */
        o += write_osc_float(buffer + o, wax9Packet->mag.x * 0.10f);            /* [OSC float] Mag.   X-axis <4 bytes> */
        o += write_osc_float(buffer + o, wax9Packet->mag.y * 0.10f);            /* [OSC float] Mag.   Y-axis <4 bytes> */
        o += write_osc_float(buffer + o, wax9Packet->mag.z * 0.10f);            /* [OSC float] Mag.   Z-axis <4 bytes> */
````

It would be great if these parameters can be changed without recompiling or adding multiply blocks in the host.

Thanks

12deadpixels, 2016-03-25:
Rate cannot be changed too.

danielgjackson, 2016-09-16:
waxrec now allows a parameter: -scale <accel> <gyro> <mag>
...to override the (multiplactive) scaling factor.

The user must ensure the matching settings (and the rate value) are sent to the device (e.g. in the -init string).

closed 2016-09-16 by danielgjackson