openmovementproject/openmovement #24: How to calculate upper arm angel and angular velocity

Annettdalboege, 2017-01-20:
Hi
I´ve just started using Axivity (AX3) and OmGui. Does anyone know how to calculate upper arm elevation and angular velocity from data :) 
Also - I would very much like to read more about how to transform AX3 data into angle and velocities so I would be grateful if anyone could send a reference or link.

Kind regards
Annett

henrikenquist, 2017-01-20:
Hi Annett

We implement AX3 inclinometers in our Lund EMINGO software package. It will be included in the next version of INC. Some testing and calibration details remain.

Best regards,
Henrik Enquist

danielgjackson, 2021-07-14:
(Closing old issues) Apologies for the very late response. 

@henrikenquist, for the benefit of anyone finding this issue --  is there a link to the EMINGO software?

closed 2021-07-14 by danielgjackson