openmovementproject/openmovement #25: Where to but WAX9? No mention of it on Axivity.com

achoo486, 2017-08-20:
I need to buy WAX9 because of its Bluetooth and gyro. But there is no mention of it on Axivity website (axivity.com). Can anyone tell me why is it so and where to buy it from?

rtmarcotte, 2017-12-12:
I inquired about it myself, here's the response from their PR:

"Unfortunately, we are no longer manufacturing the WAX9 sensor.  We are aware of the Metalab devices (although have no experience of them) which maybe of interest to you: https://mbientlab.com/"

Hope this helps!

TZebin, 2018-06-25:
Hi, is the design and schematic files are available for reproducing WAX9. I was looking into WAX9.1.DSN in OrCAD,  it says ISIS schematic file. As you have the sources, it would be good to know what software had been used while creating it.

Thanks!

danielgjackson, 2021-07-14:
(Closing old issues) I'm afraid I don't have additional information about this.

closed 2021-07-14 by danielgjackson