openmovementproject/openmovement #31: Absolute path of .cwa file not working under UNIX systems.

NarayanSchuetz, 2019-02-01:
Just wanted to let it be known here, if anyone is having the same issue and looking for a an answer:

The `c` cwa-converter does not work when the absolute path to the `.cwa` file is declare in the CLI with UNIX systems. The reason for this behaviour is the forward slash that is parsed because of the windows version. https://github.com/digitalinteraction/openmovement/blob/77c829a1163acc2404814bb48c9599caa3e2201b/Software/AX3/cwa-convert/c/main.c#L902

If I find the time I will submit a proper fix but the quick fix if you work on a UNIX system is to just remove the forward slash part (since there is no use for it beyond Windows anyways):

    if (argv[i][0] == '-')

adamcatto, 2021-07-13:
I had the same issue and for some reason it didn't go away when I removed the forward slash part; however, I've written a pretty straightforward Python function to automatically modify the full path into an equivalent relative path. You just need to write the proper path to the cwa-convert build in `CWA_CONVERT_CMD_PATH`:

```python 
import os
import subprocess

# note: first, openmovement git repo has to be cloned, then the C library must be built.

CWA_CONVERT_CMD_PATH = '/path/to/openmovement/Software/AX3/cwa-convert/c/cwa-convert'
# note: C code has a bug where it can't read an argument starting with '/' as a file
#       so we need to specify the number of "../" prefixes to put...

def convert_cwa_to_csv(input_filename, output_filename):
    # make sure that we have gotten full path to input_file
    if input_filename[0] != '/':
        # TODO{try to handle non-full-paths by using glob.iglob and inferring the proper path.}
        raise Exception('Must specify full path. i.e. /path/to/file.ext')
    filename = input_filename.split('/')[-1]
    pwd = os.getcwd()
    num_directory_levels_up_to_go = len(pwd.split('/')) - 2
    input_dir_prefix = '../' * num_directory_levels_up_to_go
    full_input_path = input_dir_prefix + filename

    subprocess.run([CWA_CONVERT_CMD_PATH, full_input_path, '-out', output_filename])
```

danielgjackson, 2021-07-14:
Sorry this took so long to fix. Thanks for the original report @NarayanSchuetz and recent comment @adamcatto that notified me that this was still an issue.  Fixed in a82452f4f98d0a408b8bd7755edc4ebdfe593d0b

closed 2021-07-14 by danielgjackson