openmovementproject/openmovement #32: Not enough data for this WTV_sd method Sampling frequency must be finite.!!

halz2656, 2019-03-20:
Hi everyone,
I just faced this problem with some few files. When I try to extract the data it gives me the following error:
"
>> myPA = findPA('29835_0504062018.cwa','site',1,'side',1,'mass',68,'bodyFat',0.2,'epoch','days','csv','29835_0504062018.csv');
progress: 0 percent 
current path: C:\Users\halz2656\Desktop\ALL FILES\My PhD projects\3) RCT Study\Data collection\Axivity and Fitbit\Data\Axivity_Exported data\Axivity_Baseline\OmGui_Axivity_baseline_ copy
Status: Calculating the PA data.

progress: 1 percent 
progress: 1 percent 
progress: 2 percent 
progress: 2 percent 
progress: 3 percent 
progress: 4 percent 
progress: 4 percent 
progress: 5 percent 
progress: 5 percent 
progress: 6 percent 
progress: 7 percent 
progress: 7 percent 
progress: 8 percent 
progress: 8 percent 
progress: 9 percent 
Not enough data for this WTV_sd method
Sampling frequency must be finite.
Error using vertcat
Dimensions of arrays being concatenated are not consistent.

Error in findPA (line 195)
        PAdata.Raw = [PAdata.Raw; calcPA(filename, p.Results,fileinfo,startTime(i),stopTime(i))];
"

Could someone help me with that?

Thanks
Hosam

halz2656, 2019-03-30:
Could anyone help me with that, please?

danielgjackson, 2021-07-14:
(Closing old issues) Apologies for the very late response. 

If this is useful for anyone else finding this issue in the future:

I believe you are using the [findPA.m](https://raw.githubusercontent.com/digitalinteraction/openmovement/master/Software/ThirdParty/OMPA/findPA.m) file, but I am afraid I'm not able to help much with your issue.  

Something that may be relevant: that old code uses "AX3_readfile.m", which only supports AX3 recording in "packed" mode.  If you are using "unpacked" mode (or an AX6), you would need to convert the code to use [CWA_readFile.m](https://raw.githubusercontent.com/digitalinteraction/openmovement/master/Software/Analysis/Matlab/CWA_readFile.m) instead.

A popular alternative choice for processing wrist-based raw accelerometry is [GGIR](https://cran.r-project.org/web/packages/GGIR/index.html), and there is [a good introduction to the package](https://journals.humankinetics.com/view/journals/jmpb/2/3/article-p188.xml) available, and a [getting started guide](https://cran.r-project.org/web/packages/GGIR/vignettes/GGIR.html).

closed 2021-07-14 by danielgjackson