openmovementproject/openmovement #33: Cannot connect to AX3 device in AX3 GUI but OK in firmware update

taldcroft, 2019-06-02:
I am unable to get the AX3 device to show up in the AX3 GUI (Open Movement 1.0.0.37) devices table.  The installation of the software and drivers all went smoothly and the tool is running just fine.  I am running a fresh install of Windows 10 in Mac Parallels.  

When I plug in the device to USB it shows up in Ports as expected as "Microchip CDC device (COM3)".  The LED does blinking red for a few seconds, pause, blinking red, then blue, then white, then green, and finally ends up with a fading in/out white-ish led (maybe red/green/blue at once).

I downloaded and ran the AX3 firmware V47 script and it does seem to have connected with the device successfully and installed updated firmware.  Any help greatly appreciated.
```
Bootload: "CWA17_47.hex"
BOOTER

Loading .HEX file: CWA17_47.he=== 3 ranges ===
--- Range 1/3, start 0x00000000, length 0x00000200 (512) ---
--- Range 2/3, start 0x00002800, length 0x0001ce70 (118384) ---
--- Range 3/3, start 0x000557f8, length 0x00000008 (8) ---
===
x
Waiting for any devices to connect...
DEVICE ADDED: \\?\hid#vid_04d8&pid_003c#7&221b53bb&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Opening device...
Opening device: \\?\hid#vid_04d8&pid_003c#7&221b53bb&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Sending query packet...: \\?\hid#vid_04d8&pid_003c#7&221b53bb&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Getting query response...
Received query response.
Device family: 0x02
Memory region #1: type=1, address=0x00001400, length=0x00028c00 (166912)
Preparing regions from .HEX file...
Copying @0x0002a000,4...
NOTE: Partial read (0x0002a000, 4) not complete (0).

NOTE: Not copying over (read outside range?)
Erasing...
Starting erase...
Erasing...
xxxxxxxxxxxxx
Programming...
wwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwwww
Verifying...
rrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrr
NOTE: Verify OK.
Resetting...
Sending reset packet...
Closing device...
Finished
DEVICE REMOVED: \\?\hid#vid_04d8&pid_003c#7&221b53bb&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
```

taldcroft, 2019-06-02:
I just noticed there is a log with relevant information:
```
Started.
Enabling background tasks...
Current: \\Mac\Home\Documents\
Current Plugin: C:\Program Files (x86)\Open Movement\OM GUI\Plugins
UPDATE: Using address: https://raw.githubusercontent.com/digitalinteraction/openmovement/master/Downloads/AX3/omgui.ini
UPDATE: Check starting...
UPDATE: Current version 1.0.0.37 is up to date (>= 1.0.0.28)
UPDATE: Check finished.
LOG: NOTE: Initialize security result: 0x80010119
LOG: WARNING: Ignoring added device with no mount point (42643)
LOG: DEBUG: Device removed: 42643
```

danielgjackson, 2021-07-14:
(Closing old issues) Apologies for the very late response. 

I'm afraid testing for OmGUI has only been carried out on non-virtualized Windows installations, but it should still be expected to work under Parallels.

The devices should not normally need their firmware changed (although that seems to have been successful anyway).

Although I know it will be too late for your case, if someone else has a similar problem and finds this issue:

You can extract a detailed log from OmGUI about what it can see of the device:

- Ensure OmGUI is initially closed.
- Open the “Run” window (Windows key + R)
- Copy the line below and paste it in the Run box and press Enter to start OmGui with more verbose log messages (this assumes OmGui was installed in the default location):
       ```
       CMD /C "SET OMDEBUG=2 && START "" "%ProgramFiles(x86)%\Open Movement\OM GUI\OmGui.exe""
       ```
- Open the View/Log (Alt+V, L) window at the very bottom, and resize it to be a little larger
- Attach the device and wait around 10 seconds
- Click in the Log window and select all of the text and copy it to the clipboard (Ctrl+Home, Ctrl+Shift+End, Ctrl+C)
- You can then paste the log output (Ctrl+V) as further information

closed 2021-07-14 by danielgjackson