openmovementproject/openmovement #34: Discarded error, but has charge.

ttruty, 2019-12-03:
[image]
I am testing devices, and am getting this error with many during set up. They have been kept with charged and recharged every few months, but this error shows even though they still have a charge on them.
Any help is appreciated

ttruty, 2019-12-03:
So info that might help: They were charged without being to connected to pc data. Only charging cable, does this affect the time setting?

Mortengs93, 2020-06-17:
Is there any other way to solve this problem?

ricci199, 2020-12-24:
I have the same error as well after charging without PC connection.

danielgjackson, 2021-07-14:
(Closing old issues) Apologies for the very late response. For the benefit of anyone finding this in a search:

The “discharged” message is a heuristic in OmGUI, and does not necessarily mean there will be a problem.   It is triggered when the software notices a connected device had lost track of time, as this implies that the battery probably became fully discharged.  The warning is to explain that devices should not be left fully discharged for extended periods (as lithium ion batteries could become damaged if they are stored completely depleted for a significant time), and devices should be charged periodically, e.g. ~3-4 months, to ensure this doesn’t happen.  (They should automatically stop recording if the battery is low, and this should leave enough power in them for the device to be returned and downloaded – the memory is non-volatile anyway, so the data will be preserved even without any power).

It is possible to accidentally discharge a device, e.g. if left connected to a USB power source that was not providing sufficient power, such as of multiple devices are connected to a passive USB hub, or a powered hub that was no longer powered, or left attached to a computer that goes to sleep and might provide less power.

I recommend three things:

1.	**Check charging arrangement:** To double-check devices are only charged by directly connecting to a computer or a USB hub that is externally powered with its own supply (not a “passive” hub without its own power, and not left on a hub that is later removed from power or power-saves e.g. from a computer going to sleep). 

2.	**To remove the message:** the message will disappear if the device’s clock is configured to the correct time.  The easiest way to do this is configure any recording with the device.  However, the configuring software will also remembers (while it is running) which devices were discharged even if they’re disconnected/reconnected, so you might also have to restart the software too if you’re immediately plugging a device back in.

3.	**To test the device:** Where possible, I would recommend confirming that everything is ok with the device by recharging and configuring it for the recording rate, delay before recording, and recording duration you’re likely to use (even if it’s a static test in a desk drawer – perhaps for a shorter duration such as 3 days, if you don’t have much time available).  Afterwards, check that the recording was successfully completed.  If the devices record as expected, they should be expected to perform similarly in the future

closed 2021-07-14 by danielgjackson

xyangwu, 2023-01-16:
Sorry to reopen this subject. I got the same warning of *fully discharged*. Is it a message that damage to AX3 is unrecoverable?
I tried but didn't find tools to repair it or reconfigure its internal time. would appreciate some help. :)