openmovementproject/openmovement #35: Running Python script error

sajjadafridi, 2020-01-16:
Hello,
I have .cwa file I need to parse and store in SQLite DB. For this, I download the cwa.py script. I passed the input file it creates the SQLite DB file but did not store any values.
When I check out the DB file. It contains DateTime, x, y, z columns.

I tried once more by debugging the code. I concluded that after open file, it read the data in binary format which make no sense in if else condition
        self.fh = open(filename,  "rb")

Any help will be appreciated in this regard.

muschellij2, 2020-11-02:
Reproducible example:
```bash
git clone --depth=1 https://github.com/digitalinteraction/openmovement
cd openmovement/Software/AX3/cwa-convert/python
wget https://biobank.ctsu.ox.ac.uk/crystal/crystal/examples/accsamp.zip
unzip accsamp.zip
python3 cwa.py accsamp.cwa
cat accsamp.cwa.sqlite 
```

Pretty much I think the issue is that https://github.com/digitalinteraction/openmovement/blob/master/Software/AX3/cwa-convert/python/cwa.py#L117 and all other string reads fail because of a binary representation of the `header` object.  I think something like:

```python
hdr = header.decode('ascii')
```
and then
```python
if hdr == "MD":
```
would work, but I'm not a Python expert.

danielgjackson, 2021-07-14:
Apologies for the very late response. 

I believe this is fixed in the current version.

closed 2021-07-14 by danielgjackson