openmovementproject/openmovement #37: APDS-9007 light sensor range

lrq3000, 2020-05-31:
The light sensor's range is specified in [Axivity's doc](https://axivity.com/files/resources/AX3_Data_Sheet.pdf) to be 3-1000 lux. In the [UK BioBank's datasheet](http://biobank.ndph.ox.ac.uk/showcase/showcase/docs/ax3_datasheet.pdf), it's 3-300 lux. Finally, on the sensor's manufacturer datasheet ([Broadcom APDS-9007](https://docs.broadcom.com/docs/AV02-0512EN)), it is 3-70K lux.

My question is hence: which one is correct? Since it's the same sensor, I would expect the range to be 3-70K lux? Or was the sensor modified to reduce to a range of 3-1000 lux, and if so, why?

angerhang, 2021-01-01:
Yes I also came also a similar question today. What is the range that we are expecting and I get something around 70k.

danielgjackson, 2021-07-14:
Apologies for the very late response. 

The light level indicator is a little complicated.  In order to not compromise the enclosure’s protection, the sensor is used without an optical window, and so its view is diffused through the case material and is easily obscured by the wearer’s clothing etc., so does not have a simple response to direction/colour.  Additionally, as the recorded level is logarithmic, small inter-device variation will be exaggerated.  It was primarily designed to be a relative indicator of light, for example to distinguish a varying/stable, or rising/falling level, or general pattern over day/night. 

The light sensor is a Broadcom APDS-9007 ([datasheet](https://docs.broadcom.com/docs/AV02-0512EN)), the load resistor on the AX3 is 100 kOhm (converting 10uA change to 1V), and the readings are then made through a 10-bit ADC, where 1V is 341.3 ADC units, and the maximum total range of the 10-bit ADC is 0-1023 units.

closed 2021-07-14 by danielgjackson

lrq3000, 2021-07-14:
Very clear explanation, thank you very much @danielgjackson !

Given the conversion formula in [here](https://axivity.com/userguides/ax3/settings/#additional-sensors), this means the range of values in lux is: 0 to 30?

> Lux = 10(counts/341)
I assume counts is the ADC values between 0-1023 as you wrote?

Is that correct or should I ignore this equation and make my own based on calibration?